stm32duino/LSM6DSV16X #3: Qvar Data Reading

GiordanaDB, 2022-11-30:
Hi everyone, i'm trying to use this sensor with an arduino Nano 33 iot and I would like to read the "Qvar Data". I have seen in your codes there are some functions for the accelerometer data, gyroscope and so on, but none for the qvar data. I guess if it is possible to build a "Get_Qvar_Data" function similarly to the "Get_X_Axes" function present in the example. I have seen in the Reg.h file that there is a "lsm6dsv16x_ah_qvar_raw_get(lsm6dsv16x_ctx_t *ctx, int16_t *val)" however I can't understand how to replicate it on an example in order to acquire Qvar Data. I would be very grateful if someone could help me. Thank you.
Giordana

cparata, 2022-11-30:
Hello @GiordanaDB ,
Unfortunately, at the moment the driver does not support the Qvar capability. I will add it in the next months, as soon as I will have time to work on it.
Best Regards,
Carlo

GiordanaDB, 2022-11-30:
Thanks for the fast answer, unfortunately I have little time to complete my university project involving the qvar data acquisition. Is it possible to know how the "Get_X_Axes" function is built in order to replicate the same structure for a "Get_Qvar_data" function? I see that the Register.h file already contains all the configurations for the qvar data so I imagine that it's all about creating the function. Thanks a lot for the attention.
Best Regards,
Giordana

cparata, 2022-11-30:
Hello @GiordanaDB ,
if you want to contribute to add this feature, you are welcome! :-) You can get inspiration to [QVAR_Handler](https://github.com/STMicroelectronics/X-CUBE-MEMS1/blob/main/Projects/NUCLEO-F401RE/Examples/IKS01A3/DataLogExtended/Src/main.c) function that was added recently in the X-CUBE-MEMS1 package. It uses the Cube framework but you can adapt to Arduino framework just reading/writing the registers with Read_Reg/Write_Reg APIs. You can find some documentation on how to enable QVAR in LSM6DSV16X sensor [here](https://www.st.com/resource/en/application_note/an5755-qvar-sensing-channel--stmicroelectronics.pdf) in the chapter 4.
Best Regards,
Carlo

GiordanaDB, 2022-12-02:
Hello @cparata , thanks for helping. I'll let you know if there are any updates.
Good day and good work,
Giordana

closed 2023-05-18 by cparata