stm32duino/LSM6DSV16X #6: error: 'MSBFIRST' was not declared in this scope

Fehmi-Karaal, 2023-02-26:
When I test the examples in arduino IDE or platform IDE, I faced the error below,

c:\Users\karaa\Documents\Arduino\libraries\LSM6DSV16X-main\src\LSM6DSV16XSensor.h:204:58: error: 'MSBFIRST' was not declared in this scope
         dev_spi->beginTransaction(SPISettings(spi_speed, MSBFIRST, SPI_MODE3));
                                                          ^~~~~~~~
c:\Users\karaa\Documents\Arduino\libraries\LSM6DSV16X-main\src\LSM6DSV16XSensor.h:204:58: note: suggested alternative: 'SPI_MSBFIRST'
         dev_spi->beginTransaction(SPISettings(spi_speed, MSBFIRST, SPI_MODE3));
                                                          ^~~~~~~~
                                                          SPI_MSBFIRST
c:\Users\karaa\Documents\Arduino\libraries\LSM6DSV16X-main\src\LSM6DSV16XSensor.h: In member function 'uint8_t LSM6DSV16XSensor::IO_Write(uint8_t*, uint8_t, uint16_t)':
c:\Users\karaa\Documents\Arduino\libraries\LSM6DSV16X-main\src\LSM6DSV16XSensor.h:251:58: error: 'MSBFIRST' was not declared in this scope
         dev_spi->beginTransaction(SPISettings(spi_speed, MSBFIRST, SPI_MODE3));
                                                          ^~~~~~~~
c:\Users\karaa\Documents\Arduino\libraries\LSM6DSV16X-main\src\LSM6DSV16XSensor.h:251:58: note: suggested alternative: 'SPI_MSBFIRST'
         dev_spi->beginTransaction(SPISettings(spi_speed, MSBFIRST, SPI_MODE3));
                                                          ^~~~~~~~
                                                          SPI_MSBFIRST

exit status 1

fpistm, 2023-02-26:
It is a core definition:
https://github.com/stm32duino/Arduino_Core_STM32/blob/621d75921205acdeef53220897c6f815c227a3f0/cores/arduino/wiring_constants.h#L67

It is a standard Arduino definition:
https://github.com/arduino/ArduinoCore-API/blob/844e4bf20c5fad8b89a2f307ba3416f61f512eaa/api/Common.h#L29

So ESP32 core should define it.
No issue with the library.

closed 2023-02-26 by fpistm

cparata, 2023-09-22:
Hello @Fehmi-Karaal ,
if you are still interested, now the library should be compatible also with ESP32 platforms.
Best Regards,
Carlo

cparata, 2023-09-26:
Hello @Fehmi-Karaal ,
what do you mean with resetting the quaternion value? You mean to reinitialize the sensor fusion? Did you try to disable and then enable again the sensor fusion feature?
Best Regards,
Carlo

Fehmi-Karaal, 2023-09-27:
Hello Carlo,

Yes, I meant reinitializing sensor. That's what I would try, but I am currently facing a new issue. I can't read quaternion values anymore. When I print fifo_samples, it is 0. Why don't I get fifo values?

Best Regards,
Fehmi Karaalioglu

cparata, 2023-09-29:
Hello @Fehmi-Karaal ,
I guess that it is due to a wrong usage of the delay function. Please, check in your case if the delay is a positive quantity, otherwise if you read a negative quantity as unsigned long, it becomes a huge delay. We have fixed this issue in the current version of the "LSM6DSV16X_Sensor_Fusion" example that you can find in the 1.7.0 version of the library. In this new release, we have also added a new API to reset the quaternion called "Reset_SFLP".
Best Regards,
Carlo