stm32duino/LSM6DSV16X #7: no matching function for call to 'TwoWire::TwoWire(int, int)'

Fehmi-Karaal, 2023-02-28:
Hello,
When I apply the sensor API in ReadMe session. I take this error.

cparata, 2023-03-01:
Hello @Fehmi-Karaal ,
which MCU board are you using with this sensor? It seems that your core does not support the TwoWire constructor.
Best Regards,
Carlo

Fehmi-Karaal, 2023-03-01:
Esp32

cparata, 2023-03-01:
Hello @cparata ,
if you check [here](https://github.com/espressif/arduino-esp32/blob/master/libraries/Wire/src/Wire.h), you will see that ESP32 does not support the Wire constructor above. So, I suggest to use one of the available Wire instances and then set SDA and SCL pins according your needs. So you can try something like:

```
Wire.setPins(I2C_SDA, I2C_SCL);
Wire.begin();
LSM6DSV16XSensor AccGyr(&Wire);
AccGyr.begin();
AccGyr.Enable_X();  
AccGyr.Enable_G();
```

Best Regards,
Carlo

cparata, 2023-03-07:
Hello @Fehmi-Karaal ,
any update on this topic?
Best Regards,
Carlo

Fehmi-Karaal, 2023-03-07:
It looks worked. When set I2C pins I did not take the error but I have not tested it on physical sensor yet. I’ll keep you updated. Thank you for following this

Regards,
Fehmi

cparata, 2023-03-07:
Ok, meantime we can close this issue.
Best Regards,
Carlo

closed 2023-03-07 by cparata

Fehmi-Karaal, 2023-03-16:
Hi Carlo,

I can successfully upload the code to ESP32 after applying your suggestion. However, I still have an I2C communication error. I used  STEVAL-MKI227KA evaluation kit and connected SCL  SDA to ESP32's SCL and SDA pins. VD and VDDIO to 3v pin in ESP32 and GND to GND.

Do you have any suggestions?

Thank you,
Fehmi Karaalioglu

Closed #7<https://github.com/stm32duino/LSM6DSV16X/issues/7> as completed.

cparata, 2023-03-17:
Hello @Fehmi-Karaal ,
you must connect also SDO/SA0 pin. You can choose to connect this pin to VDD and in this case you must use as I2C address LSM6DSV16X_I2C_ADD_H or you can connect this pin to GND and in this case you must use as I2C address LSM6DSV16X_I2C_ADD_L.
Let me know if this hint fixes your issue.
Best Regards,
Carlo

Fehmi-Karaal, 2023-03-17:
.... Thank you so much. the hint fixed the issue when I connected SDO/SA0 to  LSM6DSV16X_I2C_ADD_H. However, the output values are only 0.
The code and output are below.

#include <LSM6DSV16XSensor.h>
#include <Wire.h>
#include <Arduino.h>
#define I2C_SDA 21
#define I2C_SCL 22

LSM6DSV16XSensor sensor (&Wire);
int32_t accel[3], angrate[3];

void setup() {
  Wire.setPins(I2C_SDA, I2C_SCL);
  Wire.begin();
  Serial.begin(115200);
  sensor.begin();
  sensor.Enable_X();
  sensor.Enable_G();
}

void loop() {
  sensor.Get_X_Axes(accel);
  sensor.Get_G_Axes(angrate);

  Serial.print("Accel-X[mg]:");
  Serial.print(accel[0], 2);
  Serial.print(",Accel-Y[mg]:");
  Serial.print(accel[0], 2);
  Serial.print(",Accel-Z[mg]:");
  Serial.print(accel[0], 2);

  Serial.print(",AngRate-X[mdps]:");
  Serial.println(angrate[0], 2);
  Serial.print(",AngRate-Y[mdps]:");
  Serial.println(angrate[1], 2);
  Serial.print(",AngRate-Z[mdps]:");
  Serial.println(angrate[2], 2);

  delay(500);
}

[cid:76d1c4ca-03ee-424c-b437-c9f9f515194d]

Thank you,
Fehmi Karaalioglu

Fehmi-Karaal, 2023-03-17:
... I restarted. Now, it works well. I just need to convert from binary to decimal.

Thank you,
Fehmi Karaalioglu

Fehmi-Karaal, 2023-06-07:
Hello Carlo,

First of all, thank you for your contributions.

 I could not find the method for accessing the Machine learning core in the source codes and examples. Could you give me some tips on how can I access the machine learning core features provided by the LSM6DSV16X Inertial Measurement Unit (IMU)? I have found a similar application on LSM6DSOX but the implementation details are not publicly available in the documentation. (https://docs.arduino.cc/tutorials/nano-rp2040-connect/rp2040-imu-advanced)

Best Regards,
Fehmi Karaalioglu

cparata, 2023-06-07:
Hello @Fehmi-Karaal ,
basically there are not differences in the usage of the MLC for LSM6DSOX and the LSM6DSV16X. You just need to load the correct bytecode for the LSM6DSV16X. You can find some examples of bytecode for the LSM6DSV16X [here](https://github.com/STMicroelectronics/STMems_Machine_Learning_Core/tree/master/application_examples/lsm6dsv16x). In particular if you want to test the activity recognition example you can use [this bytecode](https://github.com/STMicroelectronics/STMems_Machine_Learning_Core/blob/master/application_examples/lsm6dsv16x/activity_recognition_for_mobile/lsm6dsv16x_activity_recognition_for_mobile.h). The way to load the bytecode of the sensor is the same for LSM6DSOX and LSM6DSV16X. Take into account that the bytecode that I linked configures the interrupt on INT1 pin as described in the [README file](https://github.com/STMicroelectronics/STMems_Machine_Learning_Core/tree/master/application_examples/lsm6dsv16x/activity_recognition_for_mobile). So, please pay attention to connect this pin to your MCU and configure it in interrupt mode.
Best Regards,
Carlo

Fehmi-Karaal, 2023-06-08:
Thank you so much for the clarification. I just wanted to know what "Get_MLC_Output/Status" methods include.[cid:69712e45-8da1-49b5-aefd-c276f4d5f740]

Kindly,
Fehmi Karaalioglu

cparata, 2023-06-08:
Hello @Fehmi-Karaal ,
you are right, these methods are missed today in the library. I will try to add them.
Best Regards,
Carlo

Fehmi-Karaal, 2023-06-08:
> Hi @Fehmi-Karaal<https://github.com/Fehmi-Karaal> ,
> I added the missing APIs for MLC. Let me know if now you are able to run an MLC example without issues.

Hi Carlo,
Thank you for adding the missing part. I have two issues because of data types as follows in the screenshot.

[cid:756cd97e-927c-4e50-98fd-4e5fd311b7bf]

Thank you,
Fehmi Karaalioglu

cparata, 2023-06-08:
Hello @Fehmi-Karaal ,
unfortunately, I cannot see the screenshot. Which kind of issue do you have on the data types?
Best Regards,
Carlo

Fehmi-Karaal, 2023-06-08:
> ...I pasted the error messages.
> [{
>       "resource": "/c:/Users/karaa/Documents/PlatformIO/Projects/walker/src/main.cpp",
>       "owner": "cpp",
>       "severity": 8,
>       "message": "no matching function for call to 'LSM6DSV16XSensor::Get_MLC_Output(uint8_t [8])'",
>       "startLineNumber": 122,
>       "startColumn": 32,
>       "endLineNumber": 122,
>       "endColumn": 32
> }]
>
> cannot convert 'lsm6dsv16x_mlc_out_t' to 'uint8_t' {aka 'unsigned char'}

and I attached the ss.

Fehmi-Karaal, 2023-06-08:
To be more clear, What is the datatype of 'Get_MLC_Output' and 'Get_MLC_Status'. I tried both  'lsm6dsv16x_mlc_status_mainpage_t' and 'lsm6dsv16x_mlc_out_t ' AND uint8_t, but it gives following error:

[{
     
     cannot convert 'lsm6dsv16x_mlc_out_t' to 'uint8_t' {aka 'unsigned char'}
    
}]

cparata, 2023-06-09:
Hello @Fehmi-Karaal ,
with respect to the example of MLC for LSM6DSOX, for LSM6DSV16X you must use the structures defined in the APIs. If you can share also the sketch that you are using, I can help you with the modifications that should be done.
Best Regards,
Carlo

Fehmi-Karaal, 2023-06-09:
/**
 ******************************************************************************
 * @file    X_NUCLEO_IKS01A3_LSM6DSOX_MLC.ino
 * @author  SRA
 * @version V1.1.0
 * @date    March 2020
 * @brief   Arduino test application for the STMicrolectronics X-NUCLEO-IKS01A3
 *          MEMS Inertial and Environmental sensor expansion board.
 *          This application makes use of C++ classes obtained from the C
 *          components' drivers.
 ******************************************************************************
 * @attention
 *
 * <h2><center>&copy; COPYRIGHT(c) 2020 STMicroelectronics</center></h2>
 *
 * Redistribution and use in source and binary forms, with or without modification,
 * are permitted provided that the following conditions are met:
 *   1. Redistributions of source code must retain the above copyright notice,
 *      this list of conditions and the following disclaimer.
 *   2. Redistributions in binary form must reproduce the above copyright notice,
 *      this list of conditions and the following disclaimer in the documentation
 *      and/or other materials provided with the distribution.
 *   3. Neither the name of STMicroelectronics nor the names of its contributors
 *      may be used to endorse or promote products derived from this software
 *      without specific prior written permission.
 *
 * THIS SOFTWARE IS PROVIDED BY THE COPYRIGHT HOLDERS AND CONTRIBUTORS "AS IS"
 * AND ANY EXPRESS OR IMPLIED WARRANTIES, INCLUDING, BUT NOT LIMITED TO, THE
 * IMPLIED WARRANTIES OF MERCHANTABILITY AND FITNESS FOR A PARTICULAR PURPOSE ARE
 * DISCLAIMED. IN NO EVENT SHALL THE COPYRIGHT HOLDER OR CONTRIBUTORS BE LIABLE
 * FOR ANY DIRECT, INDIRECT, INCIDENTAL, SPECIAL, EXEMPLARY, OR CONSEQUENTIAL
 * DAMAGES (INCLUDING, BUT NOT LIMITED TO, PROCUREMENT OF SUBSTITUTE GOODS OR
 * SERVICES; LOSS OF USE, DATA, OR PROFITS; OR BUSINESS INTERRUPTION) HOWEVER
 * CAUSED AND ON ANY THEORY OF LIABILITY, WHETHER IN CONTRACT, STRICT LIABILITY,
 * OR TORT (INCLUDING NEGLIGENCE OR OTHERWISE) ARISING IN ANY WAY OUT OF THE USE
 * OF THIS SOFTWARE, EVEN IF ADVISED OF THE POSSIBILITY OF SUCH DAMAGE.
 *
 ******************************************************************************
 */
//NOTE: This example isn't compatible with Arduino Uno.
//NOTE: For this example you need the STEVAL-MKI197V1 board connected to the DIL24 connector of the X-NUCLEO-IKS01A3

// Includes
#include "LSM6DSV16XSensor.h"
#include "lsm6dsox_activity_recognition_for_mobile.h"
#include <Wire.h>
#include <Arduino.h>
#define I2C_SDA 21
#define I2C_SCL 22

#define INT_1 A5

//Interrupts.
volatile int mems_event = 0;

LSM6DSV16XSensor sensor (&Wire);
int32_t accel[3], angrate[3];

// MLC
ucf_line_t *ProgramPointer;
int32_t LineCounter;
int32_t TotalNumberOfLine;

void INT1Event_cb();
void printMLCStatus(uint8_t status);

void setup() {
  uint8_t mlc_out[8];
  // Led.
  pinMode(LED_BUILTIN, OUTPUT);

  // Force INT1 of LSM6DSOX low in order to enable I2C
  pinMode(INT_1, OUTPUT);

  digitalWrite(INT_1, LOW);

  delay(200);

  // Initialize serial for output.
  Serial.begin(115200);
  Wire.setPins(I2C_SDA, I2C_SCL);
  Wire.begin();
  Serial.begin(115200);
  sensor.begin();
  sensor.Enable_X();
  sensor.Enable_G();

  /* Feed the program to Machine Learning Core */
  /* Activity Recognition Default program */
  ProgramPointer = (ucf_line_t *)lsm6dsox_activity_recognition_for_mobile;
  TotalNumberOfLine = sizeof(lsm6dsox_activity_recognition_for_mobile) / sizeof(ucf_line_t);
  Serial.println("Activity Recognition for LSM6DSOX MLC");
  Serial.print("UCF Number Line=");
  Serial.println(TotalNumberOfLine);

  for (LineCounter=0; LineCounter<TotalNumberOfLine; LineCounter++) {
    if(sensor.Write_Reg(ProgramPointer[LineCounter].address, ProgramPointer[LineCounter].data)) {
      Serial.print("Error loading the Program to LSM6DSV16XSensor at line: ");
      Serial.println(LineCounter);
      while(1) {
        // Led blinking.
        digitalWrite(LED_BUILTIN, HIGH);
        delay(250);
        digitalWrite(LED_BUILTIN, LOW);
        delay(250);
      }
    }
  }

  Serial.println("Program loaded inside the LSM6DSOX MLC");

  //Interrupts.
  pinMode(INT_1, INPUT);
  attachInterrupt(INT_1, INT1Event_cb, RISING);

  /* We need to wait for a time window before having the first MLC status */
  delay(3000);

  sensor.Get_MLC_Output(mlc_out);
  printMLCStatus(mlc_out[0]);
}

void loop() {
  if (mems_event) {
    mems_event=0;
    lsm6dsv16x_mlc_status_mainpage_t  status;
    sensor.Get_MLC_Status(&status);
    if (status.is_mlc1) {
      lsm6dsv16x_mlc_out_t mlc_out[8];
      sensor.Get_MLC_Output(mlc_out);
      printMLCStatus(mlc_out[0]);
    }
  }
}

void INT1Event_cb() {
  mems_event = 1;
}

void printMLCStatus(uint8_t status) {
  switch(status) {
    case 0:
      Serial.println("Activity: Stationary");
      break;
    case 1:
      Serial.println("Activity: Walking");
      break;
    case 4:
      Serial.println("Activity: Jogging");
      break;
    case 8:
      Serial.println("Activity: Biking");
      break;
    case 12:
      Serial.println("Activity: Driving");
      break;
    default:
      Serial.println("Activity: Unknown");
      break;
  }
}

Fehmi-Karaal, 2023-06-09:
…by the way, there is no example of MLC for LSM6DSV16x on github. I imitated the code from the example of MLC for LSM6dsox

/**
 ******************************************************************************

cparata, 2023-06-09:
Hello @Fehmi-Karaal ,
try this code:

```
// Includes
#include "LSM6DSV16XSensor.h"
#include "lsm6dsv16x_activity_recognition_for_mobile.h"
#include <Wire.h>
#include <Arduino.h>
#define I2C_SDA 21
#define I2C_SCL 22

#define INT_1 A5

//Interrupts.
volatile int mems_event = 0;

LSM6DSV16XSensor sensor (&Wire);
int32_t accel[3], angrate[3];

// MLC
ucf_line_t *ProgramPointer;
int32_t LineCounter;
int32_t TotalNumberOfLine;

void INT1Event_cb();
void printMLCStatus(uint8_t status);

void setup() {
  lsm6dsv16x_mlc_out_t  mlc_out;
  // Led.
  pinMode(LED_BUILTIN, OUTPUT);

  // Force INT1 of LSM6DSOX low in order to enable I2C
  pinMode(INT_1, OUTPUT);

  digitalWrite(INT_1, LOW);

  delay(200);

  // Initialize serial for output.
  Serial.begin(115200);
  Wire.setPins(I2C_SDA, I2C_SCL);
  Wire.begin();
  Serial.begin(115200);
  sensor.begin();
  sensor.Enable_X();
  sensor.Enable_G();

  /* Feed the program to Machine Learning Core */
  /* Activity Recognition Default program */
  ProgramPointer = (ucf_line_t *)lsm6dsv16x_activity_recognition_for_mobile;
  TotalNumberOfLine = sizeof(lsm6dsv16x_activity_recognition_for_mobile) / sizeof(ucf_line_t);
  Serial.println("Activity Recognition for LSM6DSV16X MLC");
  Serial.print("UCF Number Line=");
  Serial.println(TotalNumberOfLine);

  for (LineCounter=0; LineCounter<TotalNumberOfLine; LineCounter++) {
    if(sensor.Write_Reg(ProgramPointer[LineCounter].address, ProgramPointer[LineCounter].data)) {
      Serial.print("Error loading the Program to LSM6DSV16XSensor at line: ");
      Serial.println(LineCounter);
      while(1) {
        // Led blinking.
        digitalWrite(LED_BUILTIN, HIGH);
        delay(250);
        digitalWrite(LED_BUILTIN, LOW);
        delay(250);
      }
    }
  }

  Serial.println("Program loaded inside the LSM6DSOX MLC");

  //Interrupts.
  pinMode(INT_1, INPUT);
  attachInterrupt(INT_1, INT1Event_cb, RISING);

  /* We need to wait for a time window before having the first MLC status */
  delay(3000);

  sensor.Get_MLC_Output(&mlc_out);
  printMLCStatus(mlc_out.mlc1_src);
}

void loop() {
  if (mems_event) {
    mems_event=0;
    lsm6dsv16x_mlc_status_mainpage_t  status;
    sensor.Get_MLC_Status(&status);
    if (status.is_mlc1) {
      lsm6dsv16x_mlc_out_t mlc_out;
      sensor.Get_MLC_Output(&mlc_out);
      printMLCStatus(mlc_out.mlc1_src);
    }
  }
}

void INT1Event_cb() {
  mems_event = 1;
}

void printMLCStatus(uint8_t status) {
  switch(status) {
    case 0:
      Serial.println("Activity: Stationary");
      break;
    case 1:
      Serial.println("Activity: Walking");
      break;
    case 4:
      Serial.println("Activity: Jogging");
      break;
    case 8:
      Serial.println("Activity: Biking");
      break;
    case 12:
      Serial.println("Activity: Driving");
      break;
    default:
      Serial.println("Activity: Unknown");
      break;
  }
}
```
 
pay attention that you must use the bytecode for LSM6DSV16X that you can find [here](https://raw.githubusercontent.com/STMicroelectronics/STMems_Machine_Learning_Core/master/application_examples/lsm6dsv16x/activity_recognition_for_mobile/lsm6dsv16x_activity_recognition_for_mobile.h).
Let me know if it works.
Best Regards,
Carlo

Fehmi-Karaal, 2023-06-09:
...that worked. Perfect. Thank you so much. I appreciate your time and effort. I suggest you add the repository on Git Hub so that others can benefit from that.

Best Regards,
Fehmi Karaalioglu