stm32duino/LSM6DSV16X #8: Tilt Angle

phhoef, 2023-03-30:
Hi,

I need to measure the tilt angle as precise as possible (<0,1°).
I am currently using the Murata SLC3300. This is quite convenient as is directly outputs the angle. Unfortunately, the SCL3300 is quite expensive and therefore I am looking for alternatives.
I came across the new LSM6DSV16X and your library. From what I understand the IMU does not have a built-in feature to calculate the angle, right? Would it be possible to re-use the machine learning core for that purpose?

Do you consider implementing the calculation of the tilt angle in your lib?

phhoef, 2023-03-30:
Thanks for your help. I am currently planning to use a custom ESP32 board. But maybe I should reconsider using STM32. Then I would be able to use the MotionFX lib ... Can you recommend a MCU similar to the ESP32?

cparata, 2023-03-30:
I see. Anyway, if you switch to STM32, you will have many other motion libraries like MotionTL (Tilt Sensing) that should work also in Arduino and should be the best one for your use case. I guess you refer to TouchGFX library when you talked about UI development library.
Best Regards,
Carlo

phhoef, 2023-03-30:
yes indeed, TouchGFX is the library I was referring to.
STM32 has so many solutions and libraries. It's quite overwhelming to find the sweet spot for my application, as I am not too familiar with the whole ecosystem.
The MotionTL lib looks promising. Though, it confuses me as they are talking about X-CUBE-MEMS1. Not sure what this is. 

I need also many GPIOs and this is the big downside of ESP32. STM32 controllers have way more IOs.
Maybe STM32 would be the better option... :smile:

cparata, 2023-03-30:
[X-CUBE-MEMS1](https://www.st.com/en/embedded-software/x-cube-mems1.html) package is the equivalent of the MEMS libraries that you can find in Arduino for the STM32Cube the main framework to program the STM32. So, if you decide to go to STM32, you will find in the X-CUBE-MEMS1 package all the MEMS drivers, Motion libraries and application examples for the STM32Cube environment.

phhoef, 2023-03-30:
Thanks for the explanation.
Do you know a good tutorial/overview as starting point to sort out which tools I need and who to use them?
My application is has a few components, I need to control with the MCU. It has a display, inclinometer, external device via uart, storage, camera, buzzer, battery and a few buttons.

cparata, 2023-03-30:
If you want to give a look at the Cube environment, I think that you need to download STM32CubeIDE that is free IDE to program and debug the STM32. As tutorial, you can give a look on st.com and youtube.com to find some video tutorial.

phhoef, 2023-03-30:
Thanks for your help!
I am currently using PlatformIO for many years, and was quite happy with it. Are there restrictions to the Cube libraries, when not using STM32CubeIDE?

phhoef, 2023-03-30:
thanks for your help. I'll try to find a suitable stm32 controller for my use case and give it a try 😄

fpistm, 2023-03-30:
Seems answered so I close this issue.