stm32duino/LSM6DSV16X #10: error : Float_t does not name a type

Esbriven, 2023-05-12:
Hi,
I wanted to try my STEVAL board with the LSM6DSV16X embbeded, i compiled the given example "LSM6DSV16X_DataLog_Terminal" with the arduino IDE, but it seems that the "float_t" type occurs error.
I tried to change all the occurrences in the header files "LSM6DSV16XSensor.h" and "lsm6dsv16x_reg.h" however it didn't solve the issue since, it seems that there is other errors in the "lsm6dsv16x_reg.c" source file..
Am I doing it wrong ? 
Best regards :)
[image]

cparata, 2023-05-15:
Hello @Esbriven ,
I updated the library replacing all the references to float_t with float. You can try to download the 1.2.0 release and see if all the issues are solved.
Best Regards,
Carlo

Esbriven, 2023-05-15:
Hi,
Thank you for your answer the code compile and execute now. However the data displayed on the monitor remain the same even when I'm moving my STEVAL-MKI227KA. It could be my pin connection that are incorrect though..
I connected the following :
VDD             => 3.3V pin on my  Arduino Uno R3
VDD_IO        => CS pin on my STEVAL
SDO             => GND
SCx/SDx       => GND
GND1 and 2 => GND
SDA/SCL       => SDA/SCL on the Uno

I don't know if you can help me but anyway thanks for the fast response
Best regards 

[image]

cparata, 2023-05-15:
Hello @Esbriven ,
if you connect SDO to GND, you must change the I2C address. So, you must define the sensor instance in this way:

`LSM6DSV16XSensor sensor (&Wire, LSM6DSV16X_I2C_ADD_L);`

Alternatively, if you don't want to change the code, you must connect SDO to Vdd.
Best Regards,
Carlo

closed 2023-05-15 by cparata