stm32duino/LSM6DSV16X #15: Error when getting FIFO tag

GeneralSyb, 2023-10-08:
Hi,

I'm using a custom ESP32-C3 board with the LSM6DSV16X IMU and I'm trying to get a quaternion out of the sensor using this library, however I'm running into an error. It seems like it is running into the error when it reaches the `AccGyr.FIFO_Get_Tag(&tag);` line in the sensor fusion example. When running the example code (with a few modifications to have it work correctly with my board, but that mainly has to do with pin selection) with verbose logging enabled, it says `[Wire.cpp:499] requestFrom(): i2cWriteReadNonStop returned Error -1`

Let me know if there is anything you want me to try, test, look at or show.

cparata, 2023-10-09:
Hello @GeneralSyb ,
did you try the other examples of the library like, for example, [LSM6DSV16X_DataLog_Terminal](https://github.com/stm32duino/LSM6DSV16X/tree/main/examples/LSM6DSV16X_DataLog_Terminal)? Does it work?
Best Regards,
Carlo

GeneralSyb, 2023-10-09:
Thank you for the quick response @cparata,

This time I'm getting the same error, but also `requestFrom(): i2cWriteReadNonStop returned Error 263`. I can communicate with the IMU. I've tested the device by reading and writing some of the basic registers manually and it return the expected values, but it isn't liking this library.

Ideally I'd use this library because it contains all the functions, but because of bugs (which were recently fixed) this didn't want to run on the ESP32 platform, so back then I started making my own library for just the SFLP function but got stuck on reading the FIFO register. For some reason no data was being put into the FIFO register and I couldn't figure out why. But like I said, using this library would be the best option.

cparata, 2023-10-09:
Hello @cparata ,
could you try to replace line 271 of the file LSM6DSV16XSensor.h in this way:

`dev_i2c->endTransmission(true);`

I suspect that it is an issue similar to [this one](https://github.com/stm32duino/VL53L7CX/issues/4) that we experienced with the ToF sensors on ESP32 platform.
Could you test this patch and tell me if it works on your side?
Best Regards,
Carlo

GeneralSyb, 2023-10-09:
Almost the same result. Instead of the error coming from line 499 in Wire.cpp, it is coming from line 513. I also tried without any arguments, because that is how I did it in my own code, but that had the same result. So it did something, but not much.

cparata, 2023-10-10:
Hello @GeneralSyb ,
what is your hardware setup? In particular, how is it connected the SA0 pin of LSM6DSV16X component? If it is connected to VDD, the initialization that you can find in the examples should be fine; otherwise (if SA0 pin is connected to GND) you must change the constructor of the sensor in this way:

`LSM6DSV16XSensor AccGyr(&Wire, LSM6DSV16X_I2C_ADD_L);`

I hope that it can help.
Best Regards,
Carlo

fpistm, 2023-10-10:
Maybe not related but is the ESP32 core set Pull up on each I2C lines?

GeneralSyb, 2023-10-10:
> Maybe not related but is the ESP32 core set Pull up on each I2C lines?

I have 5.1k resistors on the pullup lines. I can communicate with the IMU using my own program.

> `LSM6DSV16XSensor AccGyr(&Wire, LSM6DSV16X_I2C_ADD_L);`

When using the example I'm pulling SA0 low, because I expected that to be the default state. So I'm thinking that might be the issue. Classic read the documentation mistake🫣

cparata, 2023-10-10:
Hello @GeneralSyb ,
the default examples are done pulling the SA0 high. So, or you can change the hardware configuration pulling SA0 high and test again the examples or you must change the declaration of the LSM6DSV16X object as I showed above.
Please, let me know if it fixes your issue.
Best Regards,
Carlo

cparata, 2023-10-10:
That's great!

closed 2023-10-10 by cparata