stm32duino/LSM6DSV16X #17: Uninitialized variables cause invalid Set_x_FS behaviour

VisaJE, 2024-07-04:
I face an issue when restarting an MCU of invalid behaviour of Set_X_FS.
The problem is that Set_X_FS does not change the actual FS value as detected with Get_X_FS.

I believe the bug is caused by acc_fs and gyro_fs not being initialized in constructor nor in Enable_X or Enable_G. The result is that both variables are uninitialized at the point of checking within Set_X_FS and Set_G_FS.

The problem arises after restart, because the memory of the MCU happens to be left to the previous state, which corresponds to the
FS which I am trying to set. For the same reason it is a bit difficult to reproduce the bug platform independently.
However, it seems rather clear to me that the sensitivity variables are not correctly initialized.

cparata, 2024-07-04:
Hello @VisaJE ,
thanks to have submitted this issue. I patched the library, so now in the begin function the correct values of "acc_fs" and "gyro_fs" are set.
Let me know if it fixes your issue.
Best Regards,
Carlo

VisaJE, 2024-07-04:
Hi @cparata,
Thanks for the quick reaction!
Based on how I read the new begin() function the scale values the component are now set at the wrong line: `lsm6dsv16x_xl_full_scale_set` is called with a constant `LSM6DSV16X_4g` though acc_fs was just set to the fetched value. Perhaps `Set_X_FS` should be called instead of `lsm6dsv16x_xl_full_scale_set`, or then the Get_X_FS was meant to be called after `lsm6dsv16x_xl_full_scale_set`.

cparata, 2024-07-04:
Hi @VisaJE,
You are right. Now it should be better. :-)
Let me know if it if fine for you.
Best Regards,
Carlo

VisaJE, 2024-07-04:
Works :partying_face:

closed 2024-07-04 by VisaJE